copper-project/copper-rs #112: Multiple versions of the windows crate are used

gbin, 2024-11-25:
[image]

Investigate if we can only use one.

paulotten, 2024-12-12:
Doing a code search for `windows` I don't see it directly included as a dependency in any Cargo.toml files.

After running `cargo build` in the project root and then searching Cargo.lock in the same directory I see:

(Lines omit from each [[package]] entry for brevity.)

```
[[package]]
name = "accesskit_windows"
version = "0.20.0"
dependencies = [
 "windows 0.54.0",
]

[[package]]
name = "accesskit_windows"
version = "0.23.2"
dependencies = [
 "windows 0.58.0",
]

[[package]]
name = "gpu-allocator"
version = "0.25.0"
dependencies = [
 "windows 0.52.0",
]

[[package]]
name = "gpu-allocator"
version = "0.26.0"
dependencies = [
 "windows 0.52.0",
]

[[package]]
name = "libmacchina"
version = "7.3.1"
dependencies = [
 "windows 0.39.0",
]

[[package]]
name = "reflink-copy"
version = "0.1.20"
dependencies = [
 "windows 0.58.0",
]

[[package]]
name = "sysinfo"
version = "0.30.13"
dependencies = [
 "windows 0.52.0",
]

[[package]]
name = "wmi"
version = "0.12.2"
dependencies = [
 "windows 0.48.0",
]
```

Of these only `libmacchina` appears in a Cargo.toml file (copper-rs/components/monitors/cu_consolemon/Cargo.toml). It has this entry:

```
libmacchina = "7.3" # 8.0.0+ is not supported by pfetch at the time of writing
```

Coming back to the `windows` crate. You don't seem to be directly depending on it anywhere. It seems to a dependency of other dependencies. While you can always update your direct dependencies, I wouldn't worry about dependents of dependents unless there is an underlying security issue.

gbin, 2024-12-12:
@paulotten Thank you for looking into this. Yeah if this is too far remote there is not much we can do. I try to keep my eye on stuff like that because it bit me in the past with incompatible versions of the same crate through indirect dependencies.

closed 2024-12-12 by gbin