copper-project/copper-rs #120: Feature Request: log in debug! the copper list number

gbin, 2024-12-02:
Today we have a structured log to add arbitrary debug information in the log. It would be nice if debug! can capture the context of Copper deterministic replay (for example the copper list number) and attach it to the logged info.

Like this the debug info can be reattached to the overall execution at the message level instead of just timestamp / line-number.

makeecat, 2025-04-11:
If Copper tasks can run in parallel, how to implement it? I can only think about a thread-local approach which assumes single-threaded execution of copper lists.

gbin, 2026-01-09:
They will need to be staggered with a unique CL number anyway.

gbin, 2026-01-18:
It is hard to do without some locking or sync, and this info is in every single logged copperlist so let's skip that, too much runtime constraints for little usability gain