copper-project/copper-rs #123: Bevy 0.15 bump

gbin, 2024-12-02:
Currently the bevy_mod_picking plugin is locking us to 0.14 but it looks like picking has been upstreamed: https://bevyengine.org/news/bevy-0-15/

We would need to port rp-cu-balancebot to use the new picking system.

makeecat, 2024-12-05:
Due to https://github.com/bevyengine/bevy/issues/16562, we cannot build the current demo with bevy 0.15.0 with the feature bevy_gltf. A fix has been proposed. We will wait for bevy 0.15.1 to migrate.

gbin, 2024-12-05:
@makeecat thank you for looking into it

gbin, 2024-12-05:
Can we make it compile with the bevuy_animation feature or from the git repo on head waiting for the release? @makeecat

makeecat, 2024-12-05:
I can try to work with the git repo for the cargo.toml. I can propose a PR draft accordingly. Once bevy 0.15.1 is released, we can change the cargo.toml back.

gbin, 2024-12-05:
exactly that would be awesome!

makeecat, 2025-01-04:
I bump bevy to 0.15.1 and use bevy_picking instead of bevy_mod_picking
https://github.com/copper-project/copper-rs/pull/198

closed 2025-01-05 by makeecat