copper-project/copper-rs #124: balancebot-sim panics

stillonearth, 2024-12-03:
```
$ balancebot-sim 
$ 2023-12-03T11:06:05.260135Z  INFO bevy_render::renderer: AdapterInfo { name: "gstain", vendor: 4318, device: 9860, device_type: DiscreteGpu, driver: "", driver_info: "550.120", backend: Vulkan }
2024-12-03T11:06:05.362798Z  INFO bevy_winit::system: Creating new window "Copper Simulator" (Entity { index: 0, generation: 1 })
2024-12-03T11:06:05.363063Z  INFO winit::platform_impl::linux::x11::window: Guessed window scale factor: 2
thread 'main' panicked at /home/cwiz/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cu29-unifiedlog-0.5.0/src/lib.rs:458:29:
Failed to open file: logs/balance_0.copper
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Encountered a panic in system `balancebot_sim::setup_copper`!
2024-12-03T11:06:05.364001Z  WARN bevy_ecs::world::command_queue: CommandQueue has un-applied commands being dropped.
2024-12-03T11:06:05.364009Z  WARN bevy_ecs::world::command_queue: CommandQueue has un-applied commands being dropped.
Encountered a panic in system `bevy_app::main_schedule::Main::run_main`!
```

stillonearth, 2024-12-03:
Fixed by creating logs dir.

gbin, 2024-12-03:
Ha yes, maybe we should just put a .keep in a directory on github?