copper-project/copper-rs #143: Support mcap export format

gbin, 2024-12-10:
[mcap](https://mcap.dev/) is a standard logging format used by log visualization and management tools.

We have a placeholder for log extraction here https://github.com/copper-project/copper-rs/blob/master/core/cu29_export/src/lib.rs#L82

We would need to extend that to output a properly formed mcap file.

The rust bindings are here: https://docs.rs/mcap/latest/mcap/

eggyal, 2024-12-13:
Hi, I'm interested in tackling this one.

Having delved a little into the codebase, it appears that while the parameters of existing log entries are typed (`cu29_value::Value`), the log entries themselves are untyped/schemaless.  (Of course they do have a format into which a string rendering of each parameter is interpolated, but at best that format can only be used to determine how many parameters are expected: not what type they are/should be).

Is this a correct understanding of the status quo?

If so, unless it's possible to construct a schema from some other (out of band) source, it seems to me that:

1. we can map each distinct `CuLogEntry.msg_index` to a distinct `mcap::Channel`;
2. those channels can either be schemaless (not terribly useful) or else we can attempt to infer their schema from the log entry's format string and actual parameter values.  Unfortunately, inference might not always work:
    * suppose a parameter happens to be an *empty* `Value::Option` or `Value::Map`: it will not be possible to ascertain what the type would have been if non-empty—we could continue scanning the log until a non-empty value of that parameter is found but this could be expensive if the logs are large, and it is not guaranteed that any such parameter will ever be found
    * should, for some reason, different entries have different value types for the same parameter, any such inferred schema would be violated: is it possible that this could arise (essentially requiring some sort of union type in the schema) or would this be a fatal error?
3. Alternatively, we could add schema information to Copper's own log format (or use `mcap` internally).

Is this something that the project already has some thoughts about?  My gut says using `mcap` internally might actually be quite sensible: but obviously that'd be a much more significant change that touches on many more areas of the codebase (and requires some consideration regarding migration/ongoing support of existing deployments/log files).

gbin, 2024-12-13:
Hey @eggyal thank you for checking this thing out!

What about doing something like this: https://github.com/copper-project/copper-rs/pull/151 and asking our users to just add Schema on their messages (maybe optionally or not) so we can export it?

gbin, 2024-12-13:
using mcap internally btw is not really an option as there are allocations & jitter generation in the lib. It is also 3 to 4x slower than the pretty brutal memmap from copper :P.
Let's keep the conversion at log ingest... Here I attach the benchmark results for posterity:
```
(Initial fsync to get the pending IOs out of the way): 39.599 ms

Copper: End to End: 1.754 s
Copper: allocated: 0 bytes
Copper: mean jitter: 67 ns
Copper: max jitter: 877.055 µs

(fsync check after Copper): 9.219 ms

mcap: End to End: 5.220 s
mcap: allocated: 269583972 bytes
mcap: mean jitter: 1.097 µs
mcap: max jitter: 8.454 ms

(fsync check after Mcap): 9.811 ms
```

This is the proggy to reproduce:
```rust
use std::path::Path;
use std::sync::{Arc, Mutex};
use std::time::Duration;
use cu29::bincode::{Decode, Encode};
use cu29::prelude::*;

#[derive(Default, Clone, Debug, Encode, Decode)]
struct TestPayload {
    number: u64,
    array: [u8; 10],
    float: f64,
}

fn global_system_fsync() {
    unsafe {
        libc::sync();
    }
}

struct ScopeTimer {
    start: CuDuration,
    msg: String,
    clk: RobotClock,
}

impl ScopeTimer {
    fn new(clk: &RobotClock, msg: &str) -> Self {
        Self {
            start: clk.now(),
            msg: msg.to_string(),
            clk: clk.clone(),
        }
    }
}

impl Drop for ScopeTimer {
    fn drop(&mut self) {
        println!("{}: {}", self.msg, self.clk.now() - self.start);
    }
}

const NB_MSG: usize = 20_000_000;

fn main() {

    let _a = ScopedAllocCounter::new();

    let clk = RobotClock::new();
    {
        let _t = ScopeTimer::new(&clk, "(Initial fsync to get the pending IOs out of the way)");
        global_system_fsync();
    }

    // Copper side.
    let file_path = "./ziooon.copper";
    let mut timings = CuDurationStatistics::new(CuDuration::from(Duration::from_millis(100)));
    let copper_allocations = {
        let _t = ScopeTimer::new(&clk, "Copper: End to End");
        let UnifiedLogger::Write(logger) = UnifiedLoggerBuilder::new()
            .write(true)
            .create(true)
            .file_base_name(Path::new(file_path))
            .preallocated_size(4 * 1024 * 1024 * 1024)
            .build()
            .expect("Failed to create logger")
        else {
            panic!("Failed to create logger")
        };
        let logger = Arc::new(Mutex::new(logger));

        let allocs_before = GLOBAL.get_allocated();
        let mut stream = stream_write(logger.clone(), UnifiedLogType::CopperList, 40 * 1024 * 1024);
        {
            for i in 0..NB_MSG {
                let bf = clk.now();
                let test_payload = TestPayload {
                    number: i as u64,
                    array: [(i % 255) as u8; 10],
                    float: i as f64,
                };
                let mut msg = CuMsg::<TestPayload>::new(Some(test_payload));
                msg.metadata.process_time = PartialCuTimeRange { start: CuDuration(22).into(), end: CuDuration(23).into() };
                msg.metadata.tov = CuDuration(19).into();
                stream.log(&msg).unwrap();
                timings.record(clk.now() - bf);
            }
        }
        // here everything will be dropped and a final flush to disk will be done
        GLOBAL.get_allocated() - allocs_before
    };
    println!("Copper: allocated: {} bytes", copper_allocations);
    println!("Copper: mean jitter: {}", timings.jitter_mean());
    println!("Copper: max jitter: {}", timings.jitter_max());
    {
        // To verify that really, everything was flushed to disk, it should be close to immediate
        let _t = ScopeTimer::new(&clk, "(fsync check after Copper)");
        global_system_fsync();
    }

    let mut timings = CuDurationStatistics::new(CuDuration::from(Duration::from_millis(100)));

    // mcap side
    let mcap_allocations = {
        let _t = ScopeTimer::new(&clk, "Mcap: End to End");
        use mcap::{Channel, records::MessageHeader, Writer};
        use std::{collections::BTreeMap, fs, io::BufWriter};

        let mut out = Writer::new(
            BufWriter::new(fs::File::create("out.mcap").unwrap())
        ).unwrap();

        let my_channel = Channel {
            topic: String::from("cool stuff"),
            schema: None,
            message_encoding: String::from("application/octet-stream"),
            metadata: BTreeMap::default()
        };

        let channel_id = out.add_channel(&my_channel).unwrap();
        let mut buffer = [0u8; 1024];

        let allocs_before = GLOBAL.get_allocated();
        for i in 0..NB_MSG {
            let bf = clk.now();
            let data = TestPayload {
                number: i as u64,
                array: [(i % 255) as u8; 10],
                float: i as f64,
            };
            // apple to apple with Copper
            let length = bincode::encode_into_slice(
                data,
                &mut buffer,
                bincode::config::standard()
            ).unwrap();

            out.write_to_known_channel(
                &MessageHeader {
                    channel_id,
                    sequence: 32,
                    log_time: 23,
                    publish_time: 25
                },
                &buffer[..length]
            ).unwrap();
            timings.record(clk.now() - bf);
        }
        out.finish().unwrap();
        GLOBAL.get_allocated() - allocs_before
    };
    println!("mcap: allocated: {} bytes", mcap_allocations);
    println!("mcap: mean jitter: {}", timings.jitter_mean());
    println!("mcap: max jitter: {}", timings.jitter_max());
    {
        // To verify that really, everything was flushed to disk, it should be close to immediate
        let _t = ScopeTimer::new(&clk, "(fsync check after Mcap)");
        global_system_fsync();
    }

}
```

gbin, 2024-12-22:
@eggyal if you have a draft of the PR feel free to push it and we can jam on it

eggyal, 2024-12-23:
PS: it's likely that some of the documentation/comments are outdated.

gbin, 2024-12-23:
Cool I'll have a look! For the options we can export it as a boolean + value or something like that?

eggyal, 2024-12-23:
That was my first thought too, but it won't work because instances would always require the value be occupied: and we might not be able to construct a sane value that is safe to be ignored/discarded.

But actually we could use `BoundedArray` bounded to 1 element, such that enums lower to one such field per variant plus an integer discriminant.  👍🏻

gbin, 2025-01-22:
Where are we on this one?

gbin, 2025-03-02:
@eggyal still on that or should we pass the baton to somebody else?

gbin, 2025-09-24:
Hey @sanderpick we don't have an automatic export yet but if you want to try to whip your own, it could be very quick.

Each time you compile your copper runtime it generates a log exporter in rust you can use to extract the messages

If you already have a Copper project running.

```toml
[dependencies]
cu29-export = { workspace = true }
```

Then build a simple logexport executable first to be sure it can export your data:
```toml
[[bin]]
name = "logreader"
path = "src/logreader.rs"
```

with logreader.rs just a call to the generated code:

```rust
use cu29::prelude::*;
use cu29_export::run_cli;

gen_cumsgs!("copperconfig.ron");

fn main() {
    run_cli::<CuMsgs>().expect("Failed to run the export CLI");
}
```

That should give you the basic rust export support, check with --help on the executable for the options.

Then by replicating the loop we in the export tool do in the export you can do something like that:

```rust
let UnifiedLogger::Read(mut dl) = UnifiedLoggerBuilder::new()
        .file_base_name(&unifiedlog_base)
        .build()
        .expect("Failed to create logger reader");
let mut reader = UnifiedLoggerIOReader::new(dl, UnifiedLogType::CopperList);
let iter = copperlists_reader::<P>(&mut reader);
for entry in iter {
      // it will dump each CopperList<P> (ie. each execution loop result in a big tuple with a all the messages in it, you can use you IDE to explore the generated type with all the timestamps etc...)
}
```

Those messages are all serde compatible so you can use that to exact the field etc like a normal serde deserializer.

I hope it helps and bridge the gap until we fully implement mcap!

Ping us on discord if you encounter any snag.