copper-project/copper-rs #150: Some pre-config missing for balancebot, main panic

vsenn, 2024-12-13:
Simply running the code from read.me finished with panic
```
$ cargo install cu-rp-balancebot
$ balancebot-sim 
```

```
2024-12-13T13:53:33.522666Z  INFO bevy_render::renderer: AdapterInfo { name: "Apple M3 Max", vendor: 0, device: 0, device_type: IntegratedGpu, driver: "", driver_info: "", backend: Metal }
2024-12-13T13:53:33.572088Z  INFO bevy_winit::system: Creating new window "Copper Simulator" (Entity { index: 0, generation: 1 })
thread 'main' panicked at /.cargo/registry/src/index.crates.io-6f17d22bba15001f/cu29-unifiedlog-0.5.1/src/lib.rs:458:29:
Failed to open file: logs/balance_0.copper
stack backtrace:
   0: _rust_begin_unwind
   1: core::panicking::panic_fmt
   2: cu29_unifiedlog::make_slab_file
   3: cu29_unifiedlog::UnifiedLoggerBuilder::build
   4: cu29_helpers::basic_copper_setup
   5: core::ops::function::impls::<impl core::ops::function::FnMut<A> for &mut F>::call_mut
   6: <bevy_ecs::system::function_system::FunctionSystem<Marker,F> as bevy_ecs::system::system::System>::run_unsafe
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
Encountered a panic in system `balancebot_sim::setup_copper`!
2024-12-13T13:53:33.661180Z  WARN bevy_ecs::world::command_queue: CommandQueue has un-applied commands being dropped.
2024-12-13T13:53:33.661188Z  WARN bevy_ecs::world::command_queue: CommandQueue has un-applied commands being dropped.
Encountered a panic in system `bevy_app::main_schedule::Main::run_main`!
```

gbin, 2024-12-13:
Dang I thought it was fixed. Thanks for the report

Just create the logs directory before starting as a workaround:
```bash
mkdir logs
balancebot-sim
```

vsenn, 2024-12-13:
yep! works like a charm! Thanks

gbin, 2024-12-13:
Let's reopen this, the code is supposed to create this directory.

vsenn, 2024-12-13:
maybe a warning, not panic? let's suppose I ran in `/`? then?

gbin, 2024-12-13:
or maybe forcing that to a temp directory?

closed 2024-12-24 by gbin