copper-project/copper-rs #183: Crash logging images

gbin, 2024-12-24:
From @edgarriba

[image]

Trying: https://github.com/kornia/kornia-rs/pull/193/files

gbin, 2024-12-24:
I really wonder if the panic created the core dump or the core dump created the panic Oo

edgarriba, 2024-12-25:
To unblock myself of using copper, is it possible disable the logger ?

edgarriba, 2024-12-28:
@gbin any progress here ? I'm blocked to use copper for now

gbin, 2024-12-28:
Ok I understood what is going on.

The logger preallocates pages based on the size of the CopperList (the assumption was that what you log will be part of the copperlist) but in your case because your message do allocate outside of it, the CopperList is very small, it allocates the space for 64 for them but it is still smaller than one image.

I will put a config parameter in the ron file to override that for now to unblock us. Then with the other work I am doing on the buffer pools, I'll see if I can deduce a correct allocation size.

gbin, 2024-12-28:
@edgarriba can you try with https://github.com/copper-project/copper-rs/pull/186
adding this section to the ron file:
 logging: (
        section_size_mib: 100, // put here in Mibibytes roughly ~64x the size of everything your tasks will produce for one cycle.
    ),

Note: once we have the buffer pools finalized the runtime should be able to make this decision by itself.

edgarriba, 2024-12-29:
```bash
edgar@farmng-TensorBook-late-2021:~/software/kornia-rs$ RUST_BACKTRACE=1 RUST_LOG=info cargo run --bin copper
warning: /home/edgar/software/kornia-rs/examples/copper_pipeline/Cargo.toml: unused manifest key: dependencies.cu29.sha
warning: /home/edgar/software/kornia-rs/examples/copper_pipeline/Cargo.toml: unused manifest key: dependencies.cu29-helpers.sha
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.33s
     Running `target/debug/copper`
Webcam::start
[2024-12-29T18:17:02Z INFO  re_sdk_comms::server] New SDK client connected from: 127.0.0.1:44692
Webcam::start
RerunViz::process: logging image: ImageSize { width: 640, height: 480 }
RerunViz::process: logged image
thread 'main' panicked at /home/edgar/.cargo/git/checkouts/copper-rs-b1305cc2c7d30d00/92a180a/core/cu29_unifiedlog/src/lib.rs:111:22:
Failed to encode object in a newly minted section. Unrecoverable failure.: UnexpectedEnd
stack backtrace:
   0: rust_begin_unwind
             at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/std/src/panicking.rs:665:5
   1: core::panicking::panic_fmt
             at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/core/src/panicking.rs:74:14
   2: core::result::unwrap_failed
             at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/core/src/result.rs:1700:5
   3: core::result::Result<T,E>::expect
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/result.rs:1061:23
   4: <cu29_unifiedlog::MmapStream as cu29_traits::WriteStream<E>>::log
             at /home/edgar/.cargo/git/checkouts/copper-rs-b1305cc2c7d30d00/92a180a/core/cu29_unifiedlog/src/lib.rs:106:34
   5: cu29_runtime::curuntime::CuRuntime<CT,P,M,_>::end_of_processing::{{closure}}
             at /home/edgar/.cargo/git/checkouts/copper-rs-b1305cc2c7d30d00/92a180a/core/cu29_runtime/src/curuntime.rs:87:17
   6: core::iter::traits::iterator::Iterator::for_each::call::{{closure}}
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/traits/iterator.rs:810:29
   7: core::ops::function::impls::<impl core::ops::function::FnMut<A> for &mut F>::call_mut
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:294:13
   8: core::iter::traits::double_ended::DoubleEndedIterator::rfold
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/traits/double_ended.rs:308:21
   9: <core::iter::adapters::rev::Rev<I> as core::iter::traits::iterator::Iterator>::fold
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/adapters/rev.rs:64:9
  10: <core::iter::adapters::chain::Chain<A,B> as core::iter::traits::iterator::Iterator>::fold
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/adapters/chain.rs:123:19
  11: core::iter::traits::iterator::Iterator::for_each
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/iter/traits/iterator.rs:813:9
  12: cu29_runtime::curuntime::CuRuntime<CT,P,M,_>::end_of_processing
             at /home/edgar/.cargo/git/checkouts/copper-rs-b1305cc2c7d30d00/92a180a/core/cu29_runtime/src/curuntime.rs:79:9
  13: copper::KorniaApplication::run_one_iteration
             at ./examples/copper_pipeline/src/main.rs:6:1
  14: copper::KorniaApplication::run
             at ./examples/copper_pipeline/src/main.rs:6:1
  15: copper::main
             at ./examples/copper_pipeline/src/main.rs:24:5
  16: core::ops::function::FnOnce::call_once
             at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ops/function.rs:250:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
thread 'main' panicked at /home/edgar/.cargo/git/checkouts/copper-rs-b1305cc2c7d30d00/92a180a/core/cu29_unifiedlog/src/lib.rs:131:58:
called `Result::unwrap()` on an `Err` value: PoisonError { .. }
stack backtrace:
   0:     0x55fa33d8f1fa - std::backtrace_rs::backtrace::libunwind::trace::h5a5b8284f2d0c266
                               at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/std/src/../../backtrace/src/backtrace/libunwind.rs:116:5
   1:     0x55fa33d8f1fa - std::backtrace_rs::backtrace::trace_unsynchronized::h76d4f1c9b0b875e3
                               at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x55fa33d8f1fa - std::sys::backtrace::_print_fmt::hc4546b8364a537c6
                               at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/std/src/sys/backtrace.rs:66:9
   3:     0x55fa33d8f1fa - <std::sys::backtrace::BacktraceLock::print::DisplayBacktrace as core::fmt::Display>::fmt::h5b6bd5631a6d1f6b
                               at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/std/src/sys/backtrace.rs:39:26
   4:     0x55fa33db9003 - core::fmt::rt::Argument::fmt::h270f6602a2b96f62
                               at /rustc/90b35a6239c3d8bdabc530a6a0816f7ff89a0aaf/library/core/src/fmt/rt.rs:177:76
   5:     0x55fa33db9003 - core::fmt::write::h7550c97b06c86515
[... cut ...]
                               at /home/edgar/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/rt.rs:194:17
  36:     0x55fa336c303e - main
  37:     0x7f9371d34083 - __libc_start_main
                               at /build/glibc-LcI20x/glibc-2.31/csu/../csu/libc-start.c:308:16
  38:     0x55fa336b37fe - _start
  39:                0x0 - <unknown>
thread 'main' panicked at core/src/panicking.rs:229:5:
panic in a destructor during cleanup
thread caused non-unwinding panic. aborting.
Aborted (core dumped)
```

gbin, 2024-12-29:
That is strange, what is the size of the section you put?

edgarriba, 2024-12-29:
`const SLAB_SIZE: Option<usize> = Some(100 * 1024 * 1024);` but i don't have any reference why this numbers at all

gbin, 2024-12-29:
oh I have added a new config in the .ron file in this PR!
see the example:
```
(
    tasks: [
        (
            id: "task0",
            type: "tasks::ExampleSrc",
        ),
        (
            id: "task1",
            type: "tasks::ExampleTask",
        ),
        (
            id: "task2",
            type: "tasks::ExampleSink",
        ),
     ],
    cnx: [
        (src: "task0", dst: "task1", msg: "i32"),
        (src: "task1", dst: "task2", msg: "i32"),
    ],

    logging: (
        slab_size_mib: 1024, // Preallocates 1GiB of memory map file at a time
        section_size_mib: 100, // Preallocates 100MiB of memory map per section for the main logger.
    ),
)
```
section_size_mib <- this is the one that you need to bump up.

gbin, 2024-12-29:
FYI, a SLAB is basically the size of the memory mapped files Copper preallocates (those _0, _1... files you see need to be fixed size)
within those memory mapped files, Copper will interleave all the different streams from the system but every time will reserve some space to have some runway to write into. This is what the section_size is, basically it needs to be *at least* be able to log one loop. The issue is that originally it was designed guessing the amount to pre reserve from the size of the output of your tasks, now that we start to have data that is outside of the output (in memory pools) it guesses very wrong and the section is way too small to even log one image then freaks out.

closed 2024-12-30 by gbin