copper-project/copper-rs #195: Allow to disable the logging in the ron file

gbin, 2025-01-03:
Some applications have very limited disk space so we might want to be able to disable the logging entirely from Copper.

agatekartik, 2025-01-05:
1. Instead of a simple flag, would it be more helpful to have different configurable levels of logging and also different destinations (e.g. log to RAM instead of disk)?
2. It would be great if some more context was added

makeecat, 2025-01-05:
> 1. Instead of a simple flag, would it be more helpful to have different configurable levels of logging and also different destinations (e.g. log to RAM instead of disk)?
> 2. It would be great if some more context was added

I agree with different configurable levels: DEBUG, INFO, WARNING, ERROR, SILENT
Can you elaborate on "logging to RAM"

gbin, 2025-01-05:
ok some context:

A. There are kind of 3 types of logging I see:

1. logging the data between the tasks (to replay, analyse etc...)
2. debug logging (ie. currently this is only debug!() but we will need to add info! etc..) which is some structured text logging.
3. tasks snapshots (not implemented yet): every x amount of seconds we should call freeze on the task and record their state so it can help restoring the state. (the user might chose to be doing that only in replay BTW if they know their stack is kept fully deterministic but we can imagine usecases where users might want to do that live on the robot if they broke the determinism). 

B. But there is another level:
- the destination of that logging: today it is the unified logger logging to disk (all linearly, memory mapped etc...). At one point we can imagine small nodes with not much disk space that might need to ship all that to another node.

Today what some users are facing is that sometimes they just want to run Copper but don't care about replay or the data itself (if they use rerun remotely from a node for example) and A.1. is filling up their disk at a tune of GB/s.

Initially I was thinking at B level. ie do not log anything but maybe able to just disabling A.1. might be a good stop gap solution while people might want to have the debug info still there.

agatekartik, 2025-01-05:
then perhaps it will be a good idea to split things up into multiple (related) features each focusing on a particular aspect/advantage. e.g. the stop-gap soltution will be one, writing to nonpersistent store like RAMdisks is another, and the real one which is separating the log levels and creating an entire proper design around it will be the one (which is the longer pole)

gbin, 2025-01-05:
@agatekartik agreed, I think the PR #199 in flight with some of the changes I mention is a good one, in any case sometimes the user will want to just no task log.

On the second level maybe we can imagine an Enum you pass to the runtime with:
```rust
enum UnifiedLoggerDestination {
   DISK(path: Path), // Like it is today
   NETWORK(url: Url),  // will add later
   MEMORY,  // will add later
   NULL, // Disable entirely
}
```

AS1100K, 2025-01-08:
I would be thrilled to work on improving logging for both Level A and Level B aspects. Given the complexity and the various sub-issues involved, I believe creating a dedicated tracking issue for logging improvements would be highly beneficial. This would provide a centralized place to discuss each sub-issue in detail and ensure the efforts remain organized and focused.

closed 2025-01-08 by gbin

edgarriba, 2025-03-24:
hi! how the api looks like after this ? I want to fully disable logging as I have a running service 24h serving multi-cam streams and very quickly the system collapses because of running out of memory with hundreds of *.copper files.

gbin, 2025-03-25:
@edgarriba yes

PR #199 (2025-01-08): Make config entry to disable the logging in the ron file
Fixes #195