copper-project/copper-rs #226: Tracking iyes_perf_ui migration to bevy 0.15.1

makeecat, 2025-01-15:
## Issue
Original [iyes_perf_ui](https://github.com/IyesGames/iyes_perf_ui) has not adapt bevy 0.15.1.
## Description
We incorporate iyes_perf_ui with bevy 0.15.1 (https://github.com/copper-project/copper-rs/pull/213) by forking on a personal account.
We also submitted a PR to original crate: https://github.com/IyesGames/iyes_perf_ui/pull/24
## Action Item
Once this PR (https://github.com/IyesGames/iyes_perf_ui/pull/24) is merged or the issue is addressed, we should change the source of iyes_perf_ui back to the original crate: https://github.com/copper-project/copper-rs/blob/d5aa7e7b2e1480471d1559e99c33791e4667a87b/examples/cu_rp_balancebot/Cargo.toml#L35

gbin, 2025-01-21:
I had to remove it from the release, lets put it back once it is done.

makeecat, 2025-01-22:
relevant commits:
https://github.com/copper-project/copper-rs/commit/85c4d5ca7f6318f04a91987c8d06b42c21047ac5
https://github.com/copper-project/copper-rs/commit/e2c1b48ceccf5a096a7b4d226829f167d71dab18
https://github.com/copper-project/copper-rs/commit/9afb3d558ddd736c19643c2de95a6af6789f3753

gbin, 2025-03-02:
It looks like it happened? https://github.com/IyesGames/iyes_perf_ui/pull/22#issuecomment-2505797137

makeecat, 2025-03-02:
> It looks like it happened? IyesGames/iyes_perf_ui#22 (comment)
> <https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_IyesGames_iyes-5Fperf-5Fui_pull_22-23issuecomment-2D2505797137&d=DwMFaQ&c=slrrB7dE8n7gBJbeO0g-IQ&r=xo2bIyArK07p25wVbYxN5TqpeH3Gb1149w81RRTbVlE&m=Er57T979MpuCkbRcl9m9sAlGmYYRffINhdfqlR6eOFbMMDcHR3IkrWizmnHX7s9e&s=puxEVrBtrUWWCrYBWwXv95nl1Pc-89SUKqdmCyUgpmA&e=>

Need to wait for them to bump to 0.15.1

Best Wishes,
Yang Zhou
Ph.D. candidate in Electrical and Computer Engineering
Agile Robotics and Perception Lab
Tandon School of Engineering, New York University

PR #263 (2025-03-13): Add `iyes_perf_ui` back
Fixes #226 

The issue was blocked because `iyes_perf_ui` isn't using `0.15.1` version of bevy in it's `Cargo.toml` but instead using `0.15.0`.
This is not a real blocker for us as mentioned in [Specifying Dependencies](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-cratesio) of Cargo Guide, the version `0.15.0` is `>=0.15.0`, and `<0.16.0`.

And since, we want to use anything `>=0.15.1`, this is not going to be any issue as Cargo will automatically select the latest version which today is `0.15.3`