copper-project/copper-rs #235: Issue building "cu-rp-balancebot"

avanmalleghem, 2025-01-30:
## Issue

When `cargo install cu-rp-balancebot` on my laptop, I ran into the following issue:
```
error[E0308]: mismatched types
   --> /home/antoine/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cu-consolemon-0.6.0/src/sysinfo.rs:79:13
    |
78  |         PfetchInfo::Title => pfetch::user_at_hostname(
    |                              ------------------------ arguments to this function are incorrect
79  |             &readouts.general_readout,
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^ expected `LinuxGeneralReadout`, found a different `LinuxGeneralReadout`
    |
    = note: `LinuxGeneralReadout` and `LinuxGeneralReadout` have similar names, but are actually distinct types
note: `LinuxGeneralReadout` is defined in crate `libmacchina`
   --> /home/antoine/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libmacchina-7.3.1/src/linux/mod.rs:35:1
    |
35  | pub struct LinuxGeneralReadout {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: `LinuxGeneralReadout` is defined in crate `libmacchina`
   --> /home/antoine/.cargo/registry/src/index.crates.io-6f17d22bba15001f/libmacchina-8.0.0/src/linux/mod.rs:35:1
    |
35  | pub struct LinuxGeneralReadout {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    = note: perhaps two different versions of crate `libmacchina` are being used?
note: function defined here
   --> /home/antoine/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pfetch-2.11.1/src/lib.rs:137:8
    |
137 | pub fn user_at_hostname(
    |        ^^^^^^^^^^^^^^^^
```

I understand this is a libmacchina versions mismatch because cu-consolemon tries to use both v7.3.1 and v8.0.0 but I'm not sure how to solve it the clean way.

## Setup

- ubuntu 22.04
- x86_64
- rustc 1.84.0

makeecat, 2025-01-30:
Hi, could you try to build it from source and run it from source? It is passing the CI currently, and I can work with that on my laptop with the same setup.

There may be an issue with the release on the crates.io

gbin, 2025-01-30:
A dependency was not completely pinned down and 2 versions of libmachina got brought in. Fix on its way.

gbin, 2025-01-30:
@makeecat yes, the fix is already in master, It didn't hit me that it would brake the release after the fact: 56d8388 2025-01-23 Guillaume Binet       pfetch finally decided to update their dependencies (#233)   

making a patch release just for cu_consolemon

gbin, 2025-01-30:
@avanmalleghem 
The fix is up:
cargo update
cargo install cu-rp-balancebot
balancebot-sim

# should work

avanmalleghem, 2025-01-30:
Works ! Thank you

closed 2025-01-30 by avanmalleghem