copper-project/copper-rs #236: Weird refresh issue with Consolemon

gbin, 2025-01-30:
changing screens do not refresh the content, if you click in the terminal it does refresh.

I am not sure if this is an issue only on my setup or not:

arch linux cachyOS
kitty terminal
zellij multiplexer

makeecat, 2025-01-30:
On my setup, it requires two keyboard press / one keyboard press + mouse input (move the mouse or click the mouse) in order to switch panel / refresh the panel

Ubuntu 22.04
bash/zsh terminal

gbin, 2025-01-30:
Ok this is definitely a regression... Let me check that out.

gbin, 2025-01-30:
Confirmed, it is the debug pane:

cu-consolemon = { path = "../../components/monitors/cu_consolemon", default-features = false, version = "0.6" }  # WORKS

cu-consolemon = { path = "../../components/monitors/cu_consolemon", version = "0.6" }  # BUGGY

@AS1100K for info, I am looking into it.

gbin, 2025-01-30:
@AS1100K with the debug pane there was 2 places in the loop where we were consuming events so it was ignoring keystrokes.

@makeecat if you confirm it is fixed on your side I will push another hot fix for the release, thanks!

makeecat, 2025-01-30:
confirmed, it is fixed

gbin, 2025-02-03:
fixed thanks

closed 2025-02-03 by gbin