copper-project/copper-rs #253: Build is broken due to bincode-related errors

spease, 2025-03-07:
The balancebot sim or simply building the workspace (`cargo check`) fails due to a bincode-related error. From cursory troubleshooting, this looks like it's probably because bincode just released the update to 2.0 - however pinning to "=2.0.0-rc3" in the workspace Cargo.toml still doesn't fix it. 

```
error[E0107]: trait takes 0 generic arguments but 1 generic argument was supplied
  --> core/cu29_traits/src/lib.rs:70:19
   |
70 | #[derive(dEncode, dDecode, Copy, Clone, Debug, PartialEq)]
   |                   ^^^^^^^ expected 0 generic arguments
   |
note: trait defined here, with 0 generic parameters
  --> /home/spease/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bincode-2.0.0-rc.3/src/de/mod.rs:85:11
   |
85 | pub trait Decode: Sized {
   |           ^^^^^^
   = note: this error originates in the derive macro `dDecode` (in Nightly builds, run with -Z macro-backtrace for more info)

   Compiling icu_normalizer v1.5.0
error[E0207]: the type parameter `__Context` is not constrained by the impl trait, self type, or predicates
  --> core/cu29_traits/src/lib.rs:70:19
   |
70 | #[derive(dEncode, dDecode, Copy, Clone, Debug, PartialEq)]
   |                   ^^^^^^^ unconstrained type parameter
   |
   = note: this error originates in the derive macro `dDecode` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0107]: trait takes 0 generic arguments but 1 generic argument was supplied
  --> core/cu29_traits/src/lib.rs:70:19
   |
70 | #[derive(dEncode, dDecode, Copy, Clone, Debug, PartialEq)]
   |                   ^^^^^^^ expected 0 generic arguments
   |
note: trait defined here, with 0 generic parameters
  --> /home/spease/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bincode-2.0.0-rc.3/src/de/mod.rs:95:11
   |
95 | pub trait BorrowDecode<'de>: Sized {
   |           ^^^^^^^^^^^^
   = note: this error originates in the derive macro `dDecode` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0220]: associated type `Context` not found for `Decoder`
  --> core/cu29_traits/src/lib.rs:70:19
   |
70 | #[derive(dEncode, dDecode, Copy, Clone, Debug, PartialEq)]
   |                   ^^^^^^^ associated type `Context` not found
   |
   = note: this error originates in the derive macro `dDecode` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0220]: associated type `Context` not found for `BorrowDecoder`
  --> core/cu29_traits/src/lib.rs:70:19
   |
70 | #[derive(dEncode, dDecode, Copy, Clone, Debug, PartialEq)]
   |                   ^^^^^^^ associated type `Context` not found
   |
   = note: this error originates in the derive macro `dDecode` (in Nightly builds, run with -Z macro-backtrace for more info)
```

gbin, 2025-03-08:
Thanks let me check that you... I believe you cannot pin on -something with Cargo.

gbin, 2025-03-08:
fixed on master

closed 2025-03-08 by gbin