copper-project/copper-rs #342: Setting Source Task Frequency

N8BWert, 2025-05-20:
Hello all, longtime watcher who finally has time on their hands to play around with things.  I want to throw together a small pet project that involves detecting apriltags via a v4l stream, doing a bit of processing, and sending commands to esp32 controlled robots.  I have no issues with the actual v4l stream or the apriltag detection (copper provides a nice task for each of these things, thank you guys for that).  I just wanted to know if there is currently a way I can specify the frequency of a task running.  Specifically, I wanted to know whether I could specify the v4l stream source to only be pulled by the runtime at 30Hz (or 60Hz).  In the documentation for `CuSrcTask`, it says to set the frequency of the pulls of a src see the runtime configuration.  In the `CuConfig`'s graph, context I see a msg, batch, and store but nothing related to frequency.

Thank you guys for the help.

gbin, 2025-05-20:
Hey @N8BWert, 
By default Copper just goes through the DAG as fast as it can (it is not really a pub/sub behind the scene ;))

Option 1. As a quick fix can you try to just change the fps of your source?

```bash
v4l2-ctl --list-formats-ext -d /dev/video0
```

To get the list a available values (the Discrete entries)

And as it in your source:
```RON
tasks: [
        (
            id: "src",
            type: "cu_v4l::V4l",
            params: {
                device: 0, // /dev/video0
                width: 3280,
                height: 2160,
                fps: 30,  // frames per second, fractions are not supported yet.
                fourcc: "NV12", // format of the image
                buffers: 4, // How many images copper is able to keep in memory
                timeout_ms: 500, // How long should we wait for a new image
            },
        ),
    ]
```

Option 2. Use a gstreamer source and ask gstreamer to rate the input with videorate
```RON
    tasks: [
        (
            id: "src",
            type: "cu_gstreamer::CuDefaultGStreamer", 
            config: {  
                "pipeline": "v4l2src device=/dev/video0 ! videoconvert ! videorate ! video/x-raw, format=NV12, framerate=10/1 ! appsink name=copper",
                "caps": "video/x-raw, format=NV12, width=1920, height=1080",
            },
        ),
     ],
```

Option 3. If gstreamer is a pain on your setup, you can make a quick rate copper tasks (maybe we could add that to the standard set of tasks) along those lines:

```rust
use cu29::prelude::*;
use bincode::{Decode, Encode};
use std::marker::PhantomData;

pub struct CuRateLimiter<T>
where
    T: for<'a> CuMsgPayload + 'static,
{
    _marker: PhantomData<T>,
    interval: CuDuration,
    last_tov: Option<CuTime>,
}

impl<'cl, T> CuTask<'cl> for CuRateLimiter<T>
where
    T: CuMsgPayload + 'cl,
{
    type Input = input_msg!('cl, T);
    type Output = output_msg!('cl, T);

    fn new(config: Option<&ComponentConfig>) -> CuResult<Self> {
        let hz = config
            .and_then(|cfg| cfg.get::<f64>("rate"))
            .ok_or("Missing required 'rate' config for CuRateLimiter")?;
        let interval_ns = (1e9 / hz) as u64;
        Ok(Self {
            _marker: PhantomData,
            interval: CuDuration::from(interval_ns),
            last_tov: None,
        })
    }

    fn process(
        &mut self,
        _clock: &RobotClock,
        input: Self::Input,
        output: Self::Output,
    ) -> CuResult<()> {
        let tov = match input.metadata.tov {
            Tov::Time(ts) => ts,
            _ => return Err("Expected single timestamp TOV".into()),
        };

        let allow = match self.last_tov {
            None => true,
            Some(last) => (tov - last) >= self.interval,
        };

        if allow {
            self.last_tov = Some(tov);
            if let Some(payload) = input.payload() {
                output.set_payload(payload.clone());
            } else {
                output.clear_payload();
            }
        } else {
            output.clear_payload();
        }

        Ok(())
    }
}
```

gbin, 2025-05-20:
I went on and added a quick implementation of a rate limiter, feel to try it on.

N8BWert, 2025-05-20:
Thank you so much for the help.  Once the project is complete, I'll make sure to send a picture over.

closed 2025-05-20 by N8BWert